ros/rospack #46: Many std::string implicitely converted into char *

vins31, 2015-02-16:
I have noticed that rospack.cpp contains many std::string implicitely converted into char *
eg `std::string lang` is used in 
`for(TiXmlElement* ele2 = ele->FirstChildElement(lang);`

But `FirstChildElement()` expects a char*
So it should be
`for(TiXmlElement* ele2 = ele->FirstChildElement(lang.c_str());`

Probably with g++ it is OK, but with clang (I am under OS X) no.

May I do a pull request with the appopriate calls to c_str() when std::string are implicitely converted into char * ?

dirk-thomas, 2015-02-16:
What do you mean with "with clang it is not ok"? As far as I know we can build this package perfectly fine on OS X using clang.

vins31, 2015-02-16:
I followed the source installation tutorial and during the  `./src/catkin/bin/catkin_make_isolated --install -DCMAKE_BUILD_TYPE=Release` step, clang complained on the errors I mentioned while compiling rospack.

Anyway, I am not a C++ expert, but I believe, it is dirty not to **explicitly** convert std::string into char*.

dirk-thomas, 2015-02-16:
@wjwwood Can you comment on compiling rospack with clang on OS X?

wjwwood, 2015-02-16:
I've never had this failure building on OS X, but I might have gotten warnings about it and those might have gotten elevated to errors in a newer version of clang or something.

Either way @vins31 is correct, if TinyXML doesn't provide an overloaded version of that function that takes a `std::string`, then our code should be updated. In fact I'm not sure how that ever would have worked. As I've never seen an `std::string` be implicitly converted to a `char *` when being passed like that. For example this does not work when using `%s` and `printf`. Maybe it is because `printf` requires a `const char *`, but I'm not sure.

@vins31 I think it would be useful if you could provide the output from the compile which produces the warnings or errors you are referring too.

vins31, 2015-02-17:
I have submitted a pull-request, here are the errors

```
/Users/Telequid/local/ros/ros_catkin_ws/src/rospack/src/rospack.cpp:1025:35: error: no matching member function for call to 'FirstChildElement'
    for(TiXmlElement* ele2 = ele->FirstChildElement(lang);
                             ~~~~~^~~~~~~~~~~~~~~~~
/usr/local/include/tinyxml.h:682:16: note: candidate function not viable: no known conversion from 'const std::string' (aka 'const basic_string<char, char_traits<char>, allocator<char> >') to
      'const char *' for 1st argument
        TiXmlElement* FirstChildElement( const char * _value ) {
                      ^
/usr/local/include/tinyxml.h:681:22: note: candidate function not viable: no known conversion from 'const std::string' (aka 'const basic_string<char, char_traits<char>, allocator<char> >') to
      'const char *' for 1st argument
        const TiXmlElement* FirstChildElement( const char * _value ) const;

/Users/Telequid/local/ros/ros_catkin_ws/src/rospack/src/rospack.cpp:1027:22: error: no matching member function for call to 'NextSiblingElement'
        ele2 = ele2->NextSiblingElement(lang))
               ~~~~~~^~~~~~~~~~~~~~~~~~
/usr/local/include/tinyxml.h:665:16: note: candidate function not viable: no known conversion from 'const std::string' (aka 'const basic_string<char, char_traits<char>, allocator<char> >') to
      'const char *' for 1st argument
        TiXmlElement* NextSiblingElement( const char *_next ) {
                      ^
/usr/local/include/tinyxml.h:664:22: note: candidate function not viable: no known conversion from 'const std::string' (aka 'const basic_string<char, char_traits<char>, allocator<char> >') to
      'const char *' for 1st argument
        const TiXmlElement* NextSiblingElement( const char * ) const;

/Users/Telequid/local/ros/ros_catkin_ws/src/rospack/src/rospack.cpp:1129:37: error: no matching member function for call to 'FirstChildElement'
      for(TiXmlElement* ele2 = ele->FirstChildElement(name);
                               ~~~~~^~~~~~~~~~~~~~~~~
/usr/local/include/tinyxml.h:682:16: note: candidate function not viable: no known conversion from 'const std::string' (aka 'const basic_string<char, char_traits<char>, allocator<char> >') to
      'const char *' for 1st argument
        TiXmlElement* FirstChildElement( const char * _value ) {
                      ^
/usr/local/include/tinyxml.h:681:22: note: candidate function not viable: no known conversion from 'const std::string' (aka 'const basic_string<char, char_traits<char>, allocator<char> >') to
      'const char *' for 1st argument
        const TiXmlElement* FirstChildElement( const char * _value ) const;
                            ^

/Users/Telequid/local/ros/ros_catkin_ws/src/rospack/src/rospack.cpp:1131:24: error: no matching member function for call to 'NextSiblingElement'
          ele2 = ele2->NextSiblingElement(name))
                 ~~~~~~^~~~~~~~~~~~~~~~~~
/usr/local/include/tinyxml.h:665:16: note: candidate function not viable: no known conversion from 'const std::string' (aka 'const basic_string<char, char_traits<char>, allocator<char> >') to
      'const char *' for 1st argument
        TiXmlElement* NextSiblingElement( const char *_next ) {
                      ^
/usr/local/include/tinyxml.h:664:22: note: candidate function not viable: no known conversion from 'const std::string' (aka 'const basic_string<char, char_traits<char>, allocator<char> >') to
      'const char *' for 1st argument
        const TiXmlElement* NextSiblingElement( const char * ) const;
                            ^
/Users/Telequid/local/ros/ros_catkin_ws/src/rospack/src/rospack.cpp:1550:27: error: no matching member function for call to 'LoadFile'
  if(!stackage->manifest_.LoadFile(stackage->manifest_path_))
      ~~~~~~~~~~~~~~~~~~~~^~~~~~~~
/usr/local/include/tinyxml.h:1417:7: note: candidate function not viable: no known conversion from 'std::string' (aka 'basic_string<char, char_traits<char>, allocator<char> >') to 'TiXmlEncoding' for 1st
      argument
        bool LoadFile( TiXmlEncoding encoding = TIXML_DEFAULT_ENCODING );
             ^
/usr/local/include/tinyxml.h:1421:7: note: candidate function not viable: no known conversion from 'std::string' (aka 'basic_string<char, char_traits<char>, allocator<char> >') to 'const char *' for 1st
      argument
        bool LoadFile( const char * filename, TiXmlEncoding encoding = TIXML_DEFAULT_ENCODING );
             ^
/usr/local/include/tinyxml.h:1429:7: note: candidate function not viable: no known conversion from 'std::string' (aka 'basic_string<char, char_traits<char>, allocator<char> >') to 'FILE *'
      (aka '__sFILE *') for 1st argument
        bool LoadFile( FILE*, TiXmlEncoding encoding = TIXML_DEFAULT_ENCODING );
             ^
/Users/Telequid/local/ros/ros_catkin_ws/src/rospack/src/rospack.cpp:1601:27: error: no matching member function for call to 'IterateChildren'
  while((dep_node = root->IterateChildren(depend_tag, dep_node)))
                    ~~~~~~^~~~~~~~~~~~~~~
/usr/local/include/tinyxml.h:580:13: note: candidate function not viable: no known conversion from 'const std::string' (aka 'const basic_string<char, char_traits<char>, allocator<char> >') to
      'const char *' for 1st argument
        TiXmlNode* IterateChildren( const char * _value, const TiXmlNode* previous ) {
                   ^
/usr/local/include/tinyxml.h:579:19: note: candidate function not viable: no known conversion from 'const std::string' (aka 'const basic_string<char, char_traits<char>, allocator<char> >') to
      'const char *' for 1st argument
        const TiXmlNode* IterateChildren( const char * value, const TiXmlNode* previous ) const;
                         ^
1 warning and 6 errors generated.
make[2]: *** [CMakeFiles/rospack.dir/src/rospack.cpp.o] Error 1
make[1]: *** [CMakeFiles/rospack.dir/all] Error 2
make: *** [all] Error 2
<== Failed to process package 'rospack':
```

dirk-thomas, 2015-02-18:
Since it works on other platform as well as on OS X for other people can you please provide more information what versions of involved packages and tools you use when running into the errors: e.g. clang version, tinyxml version, any custom arguments / environment variables etc.

wjwwood, 2015-02-18:
@vins31 I appreciate the additional feedback, but I'm unable to reproduce this on my OS X machine using OS X `10.9.5`, `rospack` version `2.2.5`, and `TinyXML` version `2.6.2` from Homebrew.

wjwwood, 2015-02-18:
I think either TinyXML must have changed, because this would have never compiled, even with gcc 4.3.2 it is an error:

https://ideone.com/XfcXsK

Looking at the TinyXML source, you probably have TinyXML compiled with `TIXML_USE_STL` off:

http://sourceforge.net/p/tinyxml/git/ci/master/tree/tinyxml.h#l675

This would mean that the versions of those functions which use `std::string` are not in your header. Please give more information as to how you are building so that we can narrow down why this is not set correctly.

vins31, 2015-02-20:
I do not use Homebrew at the moment because of some conflict.

I have spotted my mistake.
I first installed TinyXml2 from the GitHub repo: https://github.com/leethomason/tinyxml2 but the header was `tinyxml2.h` so I thought I needed TinyXml1 since the required header was `tinyxml.h`.
The official version did not build any library, so I took an unofficial version: https://code.google.com/p/libtinyxml/ which seems to be way too old.

closed 2015-02-20 by vins31