ros/rospack #58: Error when crawling with permissions problems

mcamp, 2016-01-15:
As suggested by ahendrix I'm creating this issue from answer.ros.org.
There is a problem with autocompletition on folder without enough permissions.
There is a better explanation of this bug here:
http://answers.ros.org/question/224134/error-when-crawling-with-permissions-problems/

Hope I'm posting this issue in the correct place

trainman419, 2016-01-15:
I'm able to reproduce this by doing:

```
mkdir -p /tmp/test/test
chmod a-r /tmp/test
cd /tmp/test/test
rospack list-names >/dev/null
```

`rospack list-names` is one of the commands that is called from within the roscd tab-completion. I redirect stdout to /dev/null so that the error message (on stderr) is more visible.

dirk-thomas, 2016-01-15:
Thanks for reporting this. Since this is a not very common use case it is not likely that the maintainers will have time to implement this. Please consider contributing a PR to make rospack work in this scenario.

kamalpandey1993, 2018-01-02:
rospack find is giving runtime errors on my beaglebone board. Other rospack commands are running fine like rospack depends.
Here is the screenshot

[image]

dirk-thomas, 2018-01-02:
@kamalpandey1993 Your issue seems to be unrelated to this ticket. If you fill a separate ticket please make sure to include enough information, in this case e.g. the content of the file which can't be parsed. Please also don't post screen shots but copy the exact message as text in order to make it searchable.

kamalpandey1993, 2018-01-03:
@dirk-thomas apologies for the inconvenience. I've raised a new issue #86 . please have a look

dirk-thomas, 2018-01-30:
Addressed by #87.

closed 2018-01-30 by dirk-thomas