ros/rospack #61: Linking Error -- Fails to link with boost

allenh1, 2016-05-16:
In preparation for the next release, I am getting started on the new Gentoo install. I have hist the first of many walls here. I am absolutely clueless as to why this isn't building. I have Boost installed:

```
allenh1@localhost ~/ros_catkin_ws/build_isolated/rospack
$ ls /usr/lib64 | grep boost
libboost_atomic-mt.so
libboost_atomic.a
libboost_atomic.so
libboost_atomic.so.1.56.0
libboost_chrono-mt.so
libboost_chrono.a
libboost_chrono.so
libboost_chrono.so.1.56.0
libboost_container-mt.so
libboost_container.a
libboost_container.so
libboost_container.so.1.56.0
libboost_date_time-mt.so
libboost_date_time.a
libboost_date_time.so
libboost_date_time.so.1.56.0
libboost_exception.a
libboost_filesystem-mt.so
libboost_filesystem.a
libboost_filesystem.so
libboost_filesystem.so.1.56.0
libboost_graph-mt.so
libboost_graph.a
libboost_graph.so
libboost_graph.so.1.56.0
libboost_iostreams-mt.so
libboost_iostreams.a
libboost_iostreams.so
libboost_iostreams.so.1.56.0
libboost_locale-mt.so
libboost_locale.a
libboost_locale.so
libboost_locale.so.1.56.0
libboost_log-mt.so
libboost_log.a
libboost_log.so
libboost_log.so.1.56.0
libboost_log_setup-mt.so
libboost_log_setup.a
libboost_log_setup.so
libboost_log_setup.so.1.56.0
libboost_math_c99-mt.so
libboost_math_c99.a
libboost_math_c99.so
libboost_math_c99.so.1.56.0
libboost_math_c99f-mt.so
libboost_math_c99f.a
libboost_math_c99f.so
libboost_math_c99f.so.1.56.0
libboost_math_c99l-mt.so
libboost_math_c99l.a
libboost_math_c99l.so
libboost_math_c99l.so.1.56.0
libboost_math_tr1-mt.so
libboost_math_tr1.a
libboost_math_tr1.so
libboost_math_tr1.so.1.56.0
libboost_math_tr1f-mt.so
libboost_math_tr1f.a
libboost_math_tr1f.so
libboost_math_tr1f.so.1.56.0
libboost_math_tr1l-mt.so
libboost_math_tr1l.a
libboost_math_tr1l.so
libboost_math_tr1l.so.1.56.0
libboost_prg_exec_monitor-mt.so
libboost_prg_exec_monitor.a
libboost_prg_exec_monitor.so
libboost_prg_exec_monitor.so.1.56.0
libboost_program_options-mt.so
libboost_program_options.a
libboost_program_options.so
libboost_program_options.so.1.56.0
libboost_random-mt.so
libboost_random.a
libboost_random.so
libboost_random.so.1.56.0
libboost_regex-mt.so
libboost_regex.a
libboost_regex.so
libboost_regex.so.1.56.0
libboost_serialization-mt.so
libboost_serialization.a
libboost_serialization.so
libboost_serialization.so.1.56.0
libboost_signals-mt.so
libboost_signals.a
libboost_signals.so
libboost_signals.so.1.56.0
libboost_system-mt.so
libboost_system.a
libboost_system.so
libboost_system.so.1.56.0
libboost_test_exec_monitor.a
libboost_thread-mt.so
libboost_thread.a
libboost_thread.so
libboost_thread.so.1.56.0
libboost_timer-mt.so
libboost_timer.a
libboost_timer.so
libboost_timer.so.1.56.0
libboost_unit_test_framework-mt.so
libboost_unit_test_framework.a
libboost_unit_test_framework.so
libboost_unit_test_framework.so.1.56.0
libboost_wave-mt.so
libboost_wave.a
libboost_wave.so
libboost_wave.so.1.56.0
libboost_wserialization-mt.so
libboost_wserialization.a
libboost_wserialization.so
libboost_wserialization.so.1.56.0
```

And yet, this is what happens when I build attempt to build:

```
allenh1@localhost ~/ros_catkin_ws/build_isolated/rospack
$ sudo /opt/ros/kinetic/env.sh make -j8 -l1
[ 55%] Built target rospack
[ 66%] Linking CXX executable /home/allenh1/ros_catkin_ws/devel_isolated/rospack/bin/rospack
/home/allenh1/ros_catkin_ws/devel_isolated/rospack/lib/librospack.so: undefined reference to `boost::program_options::detail::cmdline::cmdline(std::vector<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, std::allocator<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > > > const&)'
/home/allenh1/ros_catkin_ws/devel_isolated/rospack/lib/librospack.so: undefined reference to `boost::program_options::arg[abi:cxx11]'
/home/allenh1/ros_catkin_ws/devel_isolated/rospack/lib/librospack.so: undefined reference to `boost::program_options::to_internal(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&)'
/home/allenh1/ros_catkin_ws/devel_isolated/rospack/lib/librospack.so: undefined reference to `boost::program_options::abstract_variables_map::operator[](std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&) const'
/home/allenh1/ros_catkin_ws/devel_isolated/rospack/lib/librospack.so: undefined reference to `boost::program_options::validate(boost::any&, std::vector<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, std::allocator<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > > > const&, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >*, int)'
/home/allenh1/ros_catkin_ws/devel_isolated/rospack/lib/librospack.so: undefined reference to `boost::program_options::options_description::options_description(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, unsigned int, unsigned int)'
/home/allenh1/ros_catkin_ws/devel_isolated/rospack/lib/librospack.so: undefined reference to `boost::program_options::value_semantic_codecvt_helper<char>::parse(boost::any&, std::vector<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, std::allocator<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > > > const&, bool) const'
collect2: error: ld returned 1 exit status
CMakeFiles/rospackexe.dir/build.make:103: recipe for target '/home/allenh1/ros_catkin_ws/devel_isolated/rospack/bin/rospack' failed
make[2]: *** [/home/allenh1/ros_catkin_ws/devel_isolated/rospack/bin/rospack] Error 1
CMakeFiles/Makefile2:264: recipe for target 'CMakeFiles/rospackexe.dir/all' failed
make[1]: *** [CMakeFiles/rospackexe.dir/all] Error 2
Makefile:138: recipe for target 'all' failed
make: *** [all] Error 2
```

Any ideas? I suspect this is some kind of issue on my end, but I figured this was worth bringing to your attention.

gerkey, 2016-05-16:
Can you give the output with `VERBOSE=1`? Presumably the boost libs are being left out of the link line for some reason.

allenh1, 2016-05-16:
Absolutely. 

```
allenh1@localhost ~/ros_catkin_ws/build_isolated/rospack $ VERBOSE=1 sudo /opt/ros/kinetic/env.sh make -j8 -l8 
[ 55%] Built target rospack
[ 77%] Linking CXX executable /home/allenh1/ros_catkin_ws/devel_isolated/rospack/bin/rosstack
[ 77%] Linking CXX executable /home/allenh1/ros_catkin_ws/devel_isolated/rospack/bin/rospack
/home/allenh1/ros_catkin_ws/devel_isolated/rospack/lib/librospack.so: undefined reference to `boost::program_options::detail::cmdline::cmdline(std::vector<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, std::allocator<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > > > const&)'
/home/allenh1/ros_catkin_ws/devel_isolated/rospack/lib/librospack.so: undefined reference to `boost::program_options::arg[abi:cxx11]'
/home/allenh1/ros_catkin_ws/devel_isolated/rospack/lib/librospack.so: undefined reference to `boost::program_options::to_internal(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&)'
/home/allenh1/ros_catkin_ws/devel_isolated/rospack/lib/librospack.so: undefined reference to `boost::program_options::abstract_variables_map::operator[](std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&) const/home/'allenh1/ros_catkin_ws/
/devel_isolated/rospack/lib/librospack.so: undefined reference to `boost::program_options::detail::cmdline::cmdline(std::vector<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, std::allocator<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > > > const&)'
/home/allenh1/ros_catkin_ws/devel_isolated/rospack/lib/librospack.so: undefined reference to `boost::program_options::arg[abi:cxx11]'
/home/allenh1/ros_catkin_ws/devel_isolated/rospack/lib/librospack.so: undefined reference to `boost::program_options::to_internal(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&)'
/home/allenh1/ros_catkin_ws/devel_isolated/rospack/lib/librospack.so: undefined reference to `boost::program_options::abstract_variables_map::operator[](std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&) const'
/home/allenh1/ros_catkin_ws/devel_isolated/rospack/lib/librospack.so: undefined reference to `boost::program_options::validate(boost::any&, std::vector<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, std::allocator<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > > > const&, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >*, int)'
/home/allenh1/ros_catkin_ws/devel_isolated/rospack/lib/librospack.so: undefined reference to `boost::program_options::options_description::options_description(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, unsigned int, unsigned int)'
/home/allenh1/ros_catkin_ws/devel_isolated/rospack/lib/librospack.so: undefined reference to `boost::program_options::value_semantic_codecvt_helper<char>::parse(boost::any&, std::vector<std::__cxx11::basic_string<char, std::char_traits<char>, std::homeallocator/allenh1/ros_catkin_ws/<devel_isolated/rospack/lib/librospack.so: undefined reference charto `boost::program_options::validate(boost::any&, std::vector<std::__cxx11::basic_string<char, std::char_traits<char>, std:> >, std::allocator<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > > > const&, bool) const'
collect2: error: ld returned 1 exit status
:allocator<char> >, std::allocator<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > > > const&, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >*, int)'
/home/allenh1/ros_catkin_ws/devel_isolated/rospack/lib/librospack.so: undefined reference to `boost::program_options::options_description::options_description(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, unsigned int, unsigned int)'
/home/allenh1/ros_catkin_ws/devel_isolated/rospack/lib/librospack.so: undefined reference to `boost::program_options::value_semantic_codecvt_helper<char>::parse(boost::any&, std::vector<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, std::allocator<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > > > const&, bool) const'
collect2: error: ld returned 1 exit status
CMakeFiles/rosstackexe.dir/build.make:103: recipe for target '/home/allenh1/ros_catkin_ws/devel_isolated/rospack/bin/rosstack' failed
make[2]: *** [/home/allenh1/ros_catkin_ws/devel_isolated/rospack/bin/rosstack] Error 1
CMakeFiles/Makefile2:301: recipe for target 'CMakeFiles/rosstackexe.dir/all' failed
make[1]: *** [CMakeFiles/rosstackexe.dir/all] Error 2
make[1]: *** Waiting for unfinished jobs....
CMakeFiles/rospackexe.dir/build.make:103: recipe for target '/home/allenh1/ros_catkin_ws/devel_isolated/rospack/bin/rospack' failed
make[2]: *** [/home/allenh1/ros_catkin_ws/devel_isolated/rospack/bin/rospack] Error 1
CMakeFiles/Makefile2:264: recipe for target 'CMakeFiles/rospackexe.dir/all' failed
make[1]: *** [CMakeFiles/rospackexe.dir/all] Error 2
Makefile:138: recipe for target 'all' failed
make: *** [all] Error 2
```

gerkey, 2016-05-16:
Hmm, still no verbose link line. Probably `sudo` is not passing through the environment variable. Try with `-E` (or `sudo bash` and then build from there).

allenh1, 2016-05-16:
Sorry about that. Got it now.

```
localhost rospack # export VERBOSE=1
localhost rospack # /opt/ros/kinetic/env.sh make -j8 -l8 
/usr/bin/cmake -H/home/allenh1/ros_catkin_ws/src/rospack -B/home/allenh1/ros_catkin_ws/build_isolated/rospack --check-build-system CMakeFiles/Makefile.cmake 0
/usr/bin/cmake -E cmake_progress_start /home/allenh1/ros_catkin_ws/build_isolated/rospack/CMakeFiles /home/allenh1/ros_catkin_ws/build_isolated/rospack/CMakeFiles/progress.marks
make -f CMakeFiles/Makefile2 all
make[1]: Entering directory '/home/allenh1/ros_catkin_ws/build_isolated/rospack'
make -f CMakeFiles/rospack.dir/build.make CMakeFiles/rospack.dir/depend
make[2]: Entering directory '/home/allenh1/ros_catkin_ws/build_isolated/rospack'
cd /home/allenh1/ros_catkin_ws/build_isolated/rospack && /usr/bin/cmake -E cmake_depends "Unix Makefiles" /home/allenh1/ros_catkin_ws/src/rospack /home/allenh1/ros_catkin_ws/src/rospack /home/allenh1/ros_catkin_ws/build_isolated/rospack /home/allenh1/ros_catkin_ws/build_isolated/rospack /home/allenh1/ros_catkin_ws/build_isolated/rospack/CMakeFiles/rospack.dir/DependInfo.cmake --color=
make[2]: Leaving directory '/home/allenh1/ros_catkin_ws/build_isolated/rospack'
make -f CMakeFiles/rospack.dir/build.make CMakeFiles/rospack.dir/build
make[2]: Entering directory '/home/allenh1/ros_catkin_ws/build_isolated/rospack'
make[2]: Nothing to be done for 'CMakeFiles/rospack.dir/build'.
make[2]: Leaving directory '/home/allenh1/ros_catkin_ws/build_isolated/rospack'
[ 55%] Built target rospack
make -f CMakeFiles/rospackexe.dir/build.make CMakeFiles/rospackexe.dir/depend
make -f CMakeFiles/rosstackexe.dir/build.make CMakeFiles/rosstackexe.dir/depend
make[2]: Entering directory '/home/allenh1/ros_catkin_ws/build_isolated/rospack'
cd /home/allenh1/ros_catkin_ws/build_isolated/rospack && /usr/bin/cmake -E cmake_depends "Unix Makefiles" /home/allenh1/ros_catkin_ws/src/rospack /home/allenh1/ros_catkin_ws/src/rospack /home/allenh1/ros_catkin_ws/build_isolated/rospack /home/allenh1/ros_catkin_ws/build_isolated/rospack /home/allenh1/ros_catkin_ws/build_isolated/rospack/CMakeFiles/rospackexe.dir/DependInfo.cmake --color=
make[2]: Entering directory '/home/allenh1/ros_catkin_ws/build_isolated/rospack'
cd /home/allenh1/ros_catkin_ws/build_isolated/rospack && /usr/bin/cmake -E cmake_depends "Unix Makefiles" /home/allenh1/ros_catkin_ws/src/rospack /home/allenh1/ros_catkin_ws/src/rospack /home/allenh1/ros_catkin_ws/build_isolated/rospack /home/allenh1/ros_catkin_ws/build_isolated/rospack /home/allenh1/ros_catkin_ws/build_isolated/rospack/CMakeFiles/rosstackexe.dir/DependInfo.cmake --color=
make[2]: Leaving directory '/home/allenh1/ros_catkin_ws/build_isolated/rospack'
make[2]: Leaving directory '/home/allenh1/ros_catkin_ws/build_isolated/rospack'
make -f CMakeFiles/rospackexe.dir/build.make CMakeFiles/rospackexe.dir/build
make -f CMakeFiles/rosstackexe.dir/build.make CMakeFiles/rosstackexe.dir/build
make[2]: Entering directory '/home/allenh1/ros_catkin_ws/build_isolated/rospack'
make[2]: Entering directory '/home/allenh1/ros_catkin_ws/build_isolated/rospack'
[ 77%] Linking CXX executable /home/allenh1/ros_catkin_ws/devel_isolated/rospack/bin/rosstack
[ 77%] Linking CXX executable /home/allenh1/ros_catkin_ws/devel_isolated/rospack/bin/rospack
/usr/bin/cmake -E cmake_link_script CMakeFiles/rosstackexe.dir/link.txt --verbose=1
/usr/bin/cmake -E cmake_link_script CMakeFiles/rospackexe.dir/link.txt --verbose=1
/usr/bin/c++      CMakeFiles/rospackexe.dir/src/rospack_main.cpp.o  -o /home/allenh1/ros_catkin_ws/devel_isolated/rospack/bin/rospack -rdynamic /home/allenh1/ros_catkin_ws/devel_isolated/rospack/lib/librospack.so -lboost_filesystem-mt -lboost_program_options-mt -lboost_system-mt -ltinyxml -lboost_filesystem-mt -lboost_program_options-mt -lboost_system-mt -lpython2.7 -Wl,-rpath,/home/allenh1/ros_catkin_ws/devel_isolated/rospack/lib: 
/usr/bin/c++      CMakeFiles/rosstackexe.dir/src/rosstack_main.cpp.o  -o /home/allenh1/ros_catkin_ws/devel_isolated/rospack/bin/rosstack -rdynamic /home/allenh1/ros_catkin_ws/devel_isolated/rospack/lib/librospack.so -lboost_filesystem-mt -lboost_program_options-mt -lboost_system-mt -ltinyxml -lboost_filesystem-mt -lboost_program_options-mt -lboost_system-mt -lpython2.7 -Wl,-rpath,/home/allenh1/ros_catkin_ws/devel_isolated/rospack/lib: 
/home/allenh1/ros_catkin_ws/devel_isolated/rospack/lib/librospack.so: undefined /home/allenh1/ros_catkin_ws/reference todevel_isolated/rospack/lib/librospack.so: undefined reference to  `boost`boost::program_options::detail::cmdline::cmdline(std::vector<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, std::allocator<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > > > const&)'
/home/allenh1/ros_catkin_ws/devel_isolated/rospack/lib/librospack.so: undefined reference to `boost::program_options::arg[abi:cxx11]'
/home/allenh1/ros_catkin_ws/devel_isolated/rospack/lib/librospack.so: undefined reference to `boost::program_options::to_internal(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> ::program_options::detail::cmdline::cmdline(std::vector<std::__cxx11::basic_string<char, std::char_traits<char>,> const&)'
/home/allenh1/ros_catkin_ws/devel_isolated/rospack/lib/librospack.so: undefined reference to `boost::program_options::abstract_variables_map::operator[](std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&) const'
[... cut ...]
make[2]: *** [/home/allenh1/ros_catkin_ws/devel_isolated/rospack/bin/rospack] Error 1
make[2]: Leaving directory '/home/allenh1/ros_catkin_ws/build_isolated/rospack'
CMakeFiles/Makefile2:264: recipe for target 'CMakeFiles/rospackexe.dir/all' failed
make[1]: *** [CMakeFiles/rospackexe.dir/all] Error 2
make[1]: Leaving directory '/home/allenh1/ros_catkin_ws/build_isolated/rospack'
Makefile:138: recipe for target 'all' failed
make: *** [all] Error 2
```

That's rather strange. Definitely appears to (attempt to) link.

allenh1, 2016-05-17:
Ehm... Worked when I installed build with Boost-1.60 -- is that the default in Ubuntu these days? Gentoo uses 1.59 unless you request otherwise.  Is this expected to fail the build?

gerkey, 2016-05-17:
Hmm, I don't know why the boost version would matter for this section of code, which is by now very old and unlikely needs a particular version of boost.

Not sure that I'm interpreting the symbol errors correctly, but: is it possible that your 1.59 of boost was compiled without c++11 support, while your 1.60 was compiled with c++11 support (e.g., perhaps you installed a c++11-compliant compiler in between)?

allenh1, 2016-05-17:
That's highly doubtful. Nowadays, Gentoo ships with GCC-4.9.3 (which I believe even has C++14 support), and the compilation of both versions of boost used "-std=gnu98" when compiling (which definitely concerned me). I'm rather confused by all this. Anyway, I think we can close this as this appears to not be a rospack issue.

gerkey, 2016-05-17:
OK, I'll close. Re-open if you can track it back to rospack.

closed 2016-05-17 by gerkey

indiancinema, 2017-06-06:
Sorry to revive a closed thread, but I am facing the same issue. I have boost compiled with c++11 support and yet I get the following error  
```

[100%] Linking CXX executable /shared/hpc/ros_catkin_ws/devel_isolated/rospack/bin/rospack
/shared/hpc/ros_catkin_ws/devel_isolated/rospack/lib/librospack.so: undefined reference to `boost::program_options::detail::cmdline::cmdline(std::vector<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, std::allocator<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > > > const&)'
/shared/hpc/ros_catkin_ws/devel_isolated/rospack/lib/librospack.so: undefined reference to `boost::program_options::arg[abi:cxx11]'
/shared/hpc/ros_catkin_ws/devel_isolated/rospack/lib/librospack.so: undefined reference to `boost::program_options::to_internal(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&)'
/shared/hpc/ros_catkin_ws/devel_isolated/rospack/lib/librospack.so: undefined reference to `boost::program_options::abstract_variables_map::operator[](std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&) const'
/shared/hpc/ros_catkin_ws/devel_isolated/rospack/lib/librospack.so: undefined reference to `boost::program_options::validate(boost::any&, std::vector<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, std::allocator<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > > > const&, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >*, int)'
/shared/hpc/ros_catkin_ws/devel_isolated/rospack/lib/librospack.so: undefined reference to `boost::program_options::options_description::options_description(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, unsigned int, unsigned int)'
/shared/hpc/ros_catkin_ws/devel_isolated/rospack/lib/librospack.so: undefined reference to `boost::program_options::value_semantic_codecvt_helper<char>::parse(boost::any&, std::vector<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, std::allocator<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > > > const&, bool) const'
collect2: error: ld returned 1 exit status
/shared/hpc/ros_catkin_ws/devel_isolated/rospack/lib/librospack.so: undefined reference to `boost::program_options::detail::cmdline::cmdline(std::vector<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, std::allocator<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > > > const&)'
/shared/hpc/ros_catkin_ws/devel_isolated/rospack/lib/librospack.so: undefined reference to `boost::program_options::arg[abi:cxx11]'
/shared/hpc/ros_catkin_ws/devel_isolated/rospack/lib/librospack.so: undefined reference to `boost::program_options::to_internal(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&)'
/shared/hpc/ros_catkin_ws/devel_isolated/rospack/lib/librospack.so: undefined reference to `boost::program_options::abstract_variables_map::operator[](std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&) const'
/shared/hpc/ros_catkin_ws/devel_isolated/rospack/lib/librospack.so: undefined reference to `boost::program_options::validate(boost::any&, std::vector<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, std::allocator<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > > > const&, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >*, int)'
/shared/hpc/ros_catkin_ws/devel_isolated/rospack/lib/librospack.so: undefined reference to `boost::program_options::options_description::options_description(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, unsigned int, unsigned int)'
/shared/hpc/ros_catkin_ws/devel_isolated/rospack/lib/librospack.so: undefined reference to `boost::program_options::value_semantic_codecvt_helper<char>::parse(boost::any&, std::vector<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, std::allocator<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > > > const&, bool) const'
collect2: error: ld returned 1 exit status
make[2]: *** [/shared/hpc/ros_catkin_ws/devel_isolated/rospack/bin/rospack] Error 1
make[1]: *** [CMakeFiles/rospackexe.dir/all] Error 2
make[1]: *** Waiting for unfinished jobs....
make[2]: *** [/shared/hpc/ros_catkin_ws/devel_isolated/rospack/bin/rosstack] Error 1
make[1]: *** [CMakeFiles/rosstackexe.dir/all] Error 2
make: *** [all] Error 2
```

dirk-thomas, 2017-06-06:
@indiancinema Since your question seems to be related to build on a specific platform (otherwise you probably wouldn't buuild `boost` yourself) I would suggest you are your question on answers.ros.org instead.

indiancinema, 2017-06-06:
Well, I had the same error when I used the system build. I was gonna ask over at answers.ros.org too, incidentally. Thanks.