ros/rospack #63: rospack depends on rosdep

mikepurvis, 2016-07-05:
There's some discussion here: https://github.com/ros/rospack/issues/5#issuecomment-12639101

And I can't actually find any instances of rosdep being needed by this code: https://github.com/ros/rospack/search?q=rosdep&type=Code

Is the dependency completely vestigial at this point? If so, can it be removed?

Our deployed robots are not development platforms; rosdep should not be necessary as a runtime dependency of ROS; we'd like to eliminate having to install and initialize it.

dirk-thomas, 2016-07-05:
Some verbs of `rospack` call the `rosdep` Python API (https://github.com/ros/rospack/search?utf8=%E2%9C%93&q=rosdep2&type=Code). So it can not be removed as a dependency.

Btw. you might not want to trust the GitHub search. It sometimes provides you only with incomplete results. E.g. in this case it didn't return you the cases where `rosdep2` was found (while you searched for `rosdep` it assumes you want "full-word-results only").

mikepurvis, 2016-07-05:
Oh blah, I thought I did check for rosdep2, but I was looking at other repos and must have gotten confused. Anyway, looks like the immediate affected methods are `isSysPackage` and `cpp_exports`, and then `computeDepsInternal`/`computeDeps`.

At a conceptual level, I can see rospack needing to be able to distinguish between catkin deps and system deps, but this still feels like a strange dependency to me, especially when it comes up at runtime— what is it that roslaunch and rosrun are doing which ultimately comes back as a complaint from rosdep about being uninitialized?

dirk-thomas, 2016-07-05:
I don't know what exactly you are invoking. I don't know of anything  in `roslaunch` which requires `rosdep`. E.g. I can run `roslaunch roscpp_tutorials talker_listener.launch` without having initialized `rosdep`. Please provide more information to narrow down which part of your cases uses rosdep.

mikepurvis, 2016-07-09:
Ah.. hmm. You're right that it isn't required for a trivial invocation. We see it on our platforms and in buildroots running unit tests. I will reopen the ticket when I get a concrete MWE for where I think there's a wrongful rosdep dependency.

closed 2016-07-09 by mikepurvis

mikepurvis, 2016-12-21:
Just to follow up here, it's an issue with plugin discovery. Similar report here:

https://github.com/ros-planning/moveit_ros/issues/154

The underlying issue is with `rospkg.get_depends_on` and is being tracked in https://github.com/ros-infrastructure/rospkg/issues/103.