ros/rospack #67: rospack picks up the wrong opencv libs

ravich2-7183, 2017-02-04:
rospack picks up the wrong opencv libs. 

I have opencv2.4.9 libs installed in `/usr/local/lib/`, an older version (2.4.8) in `/usr/lib/x86_64-linux-gnu/`. 

cmake find_package reports the correct libs
`find_package(OpenCV 2.4.9 EXACT REQUIRED)`
`message("OpenCV_INCLUDE_DIRS: ${OpenCV_INCLUDE_DIRS}") `-> /usr/local/include/opencv/
`message("OpenCV_LIBRARIES: ${OpenCV_LIBRARIES}") `->  opencv_viz;opencv_videostab; etc
`message("OpenCV_LIBRARY_DIRS: ${OpenCV_LIBRARY_DIRS}")` -> "" (_is this a problem?_)

`$ rospack libs-only-l`
gives:
... /usr/lib/x86_64-linux-gnu/libopencv_videostab.so.2.4.8 ...: etc

Does rospack use a cache? How can I clear it?

tfoote, 2017-02-06:
rospack asks the underlying packages for their exported libraries. Either my_library or one of it's dependencies are using the older version and exporting the linker flags. If you are using any package from the binary installation and it uses the Opencv, it will require 2.4.8 since that's what it's compiled against. You need to find any package using opencv and make sure that it also finds 2.4.9 as well.

ravich2-7183, 2017-02-07:
@tfoote 
Thanks for responding.

relevant contents of my manifest.xml are:
```
<package>
  <description brief="my_package">
     my_package
  </description>

  <depend package="roscpp"/>
  <depend package="tf"/>
  <depend package="sensor_msgs"/>
  <!-- <depend package="opencv2"/> -->
  <depend package="image_transport"/>
  <depend package="cv_bridge/>

</package>
```
You are right in that cv_bridge is the package brining in the opencv2.4.8 libs as revealed by:
`rospack libs-only-l cv_bridge`

(Since I'm using **ros-indigo**, I commented out the dependency on opencv2)

For now, I deleted the opencv2.4.8 libs from `/usr/lib/x86_64-linux-gnu/`, and replaced them with symlinks to the opencv2.4.9 ones in `/usr/loca/lib/`. _Is there a better way to make sure that `/opt/ros/indigo/lib/libcv_bridge.so` finds opencv2.4.9?_

tfoote, 2017-02-07:
The correct way to fix this is to recompile all packages using opencv from source with 2.4.9 on your path such that their CMake configuration stage finds the newer version. You cannot use the precompiled binary packages. The symlinking is fragile and relies on 2.4.8 and 2.4.9 being ABI compatible. And may break any other system actually using/relying on 2.4.8.

Since this isn't a rospack issue. I'm going to close this. If you need more help please ask on http://answers.ros.org

closed 2017-02-07 by tfoote