ros/rospack #73: librospack cache feature mismatches with roslib/packages.py

githubna, 2017-06-06:
1. In librospack (rospack.cpp), the cache file name is created to be ".rospack_cache.XXXXXX". While in function _read_rospack_cache() of roslib/packages.py, it expects the file to be "rospack_cache".
313    try:
314        with open(os.path.join(rospkg.get_ros_home(), 'rospack_cache')) as f:

2. In function get_pkg_dir() of roslib/packages.py, when the cache is checked (line 161), the file checked is manifest.xml. In fact, many packages now have package.xml only. Therefore, the condition to check is the existence of either manifest.xml or package.xml.
158        if package in _pkg_dir_cache:
159            dir_, rr, rpp = _pkg_dir_cache[package]
160            if rr == ros_root and rpp == ros_package_path:
161                if os.path.isfile(os.path.join(dir_, MANIFEST_FILE)):
162                    return dir_
163                else:
164                    # invalidate cache
165                    _invalidate_cache(_pkg_dir_cache)

githubna, 2017-06-06:
Make code readable:

In librospack (rospack.cpp), the cache file name is created to be ".rospack_cache.XXXXXX". While in function _read_rospack_cache() of roslib/packages.py, it expects the file to be "rospack_cache".

```
313    try:
314        with open(os.path.join(rospkg.get_ros_home(), 'rospack_cache')) as f:
```

In function get_pkg_dir() of roslib/packages.py, when the cache is checked (line 161), the file checked is manifest.xml. In fact, many packages now have package.xml only. Therefore, the condition to check is the existence of either manifest.xml or package.xml.
```
158        if package in _pkg_dir_cache:
159            dir_, rr, rpp = _pkg_dir_cache[package]
160            if rr == ros_root and rpp == ros_package_path:
161                if os.path.isfile(os.path.join(dir_, MANIFEST_FILE)):
162                    return dir_
163                else:
164                    # invalidate cache
165                    _invalidate_cache(_pkg_dir_cache)

```

dirk-thomas, 2017-06-06:
The Python function you are referring to ``roslib.packages.get_pkg_dir()` only operates on old dry packages (which have a `manifest.xml` file). For current ROS distros it is basically not being used and only exists for legacy reasons. I guess that is the reason why it has never been updated to use the cache file which has a suffix encoding the environment it belongs to.

What is the reason that you filled the ticket in the first place / what do you expect to happen?

githubna, 2017-06-07:
We tried to launch novatel ros package on an embedding system running qnx. The error shows that the novatel msg package cannot be located by calling roslib.msgs.load_by_type(). After some tracing, we noticed subprocess.Popen does not return properly for some reason on the platform (still debugging). To make the thing work for now, I started digging in the code and noticed that the package can be located using _pkg_dir_cache. Ideally, If subprocess.Popen can run properly on our platform, there is no need to fix the issue.

dirk-thomas, 2017-06-07:
Let's wait what your Popen investigation will reveal and then go from there...

githubna, 2017-06-09:
The Popen in both subprocess/subprocess32 python modules does not work properly using PIPE under QNX if the app is using threads. Need more investigation.

```
# python ./test_subprocess32.py
test_call_kwargs (__main__.ProcessTestCase) ... ok
test_call_seq (__main__.ProcessTestCase) ... ok
test_call_timeout (__main__.ProcessTestCase) ... ok
test_check_call_nonzero (__main__.ProcessTestCase) ... ok
test_check_call_zero (__main__.ProcessTestCase) ... ok
test_check_output (__main__.ProcessTestCase) ... ok
test_check_output_nonzero (__main__.ProcessTestCase) ... ok
test_check_output_stderr (__main__.ProcessTestCase) ... ok
test_check_output_stdout_arg (__main__.ProcessTestCase) ... ok
test_check_output_timeout (__main__.ProcessTestCase) ... ok
test_communicate (__main__.ProcessTestCase) ... ok
test_communicate_eintr (__main__.ProcessTestCase) ... ok
test_communicate_pipe_buf (__main__.ProcessTestCase) ... Traceback (most recent call last):
  File "./test_subprocess32.py", line 2386, in <module>
    test_main()
  File "./test_subprocess32.py", line 2382, in test_main
    test_support.run_unittest(*unit_tests)
  File "/base/usr/lib/python2.7/test/test_support.py", line 1461, in run_unittest
    _run_suite(suite)
  File "/base/usr/lib/python2.7/test/test_support.py", line 1434, in _run_suite
    result = runner.run(suite)
  File "/base/usr/lib/python2.7/unittest/runner.py", line 151, in run
    test(result)
  File "/base/usr/lib/python2.7/unittest/suite.py", line 70, in __call__
    return self.run(*args, **kwds)
  File "/base/usr/lib/python2.7/unittest/suite.py", line 108, in run
    test(result)
  File "/base/usr/lib/python2.7/unittest/suite.py", line 70, in __call__
    return self.run(*args, **kwds)
  File "/base/usr/lib/python2.7/unittest/suite.py", line 108, in run
    test(result)
  File "/base/usr/lib/python2.7/unittest/case.py", line 393, in __call__
    return self.run(*args, **kwds)
  File "/base/usr/lib/python2.7/unittest/case.py", line 329, in run
    testMethod()
  File "./test_subprocess32.py", line 734, in test_communicate_pipe_buf
    (stdout, stderr) = p.communicate(string_to_write)
  File "/opt/ros/kinetic/lib/python2.7/site-packages/subprocess32.py", line 924, in communicate
    stdout, stderr = self._communicate(input, endtime, timeout)
  File "/opt/ros/kinetic/lib/python2.7/site-packages/subprocess32.py", line 1706, in _communicate
    orig_timeout)
  File "/opt/ros/kinetic/lib/python2.7/site-packages/subprocess32.py", line 1790, in _communicate_with_poll
    self._input_offset += os.write(fd, chunk)
KeyboardInterrupt
```

githubna, 2017-12-13:
To work around the issue, I wrote a NovAtel ROS driver in C++.

closed 2017-12-13 by githubna