ros/rospack #79: Require specific versions of tinyxml2

hashworks, 2017-08-22:
I just had a problem with rospack and tinyxml2 Version 4, XML parsing failed on the `<?xml-model ...` lines.

Would it be possible to require specific versions of tinyxml2? Installing version 5 fixed the issue for me.

dirk-thomas, 2017-08-22:
Please see #77 which covers this issue. Since only version 4.0.0 and 4.0.1 have the bug we can't simply rely on version 5. E.g. version 2.x as well as 3.x are working fine too.

Which platform are you using which still has an unpatched version 4?

hashworks, 2017-08-22:
[The Debian stretch repositories offer v4.0.1.](https://packages.debian.org/search?suite=stretch&section=all&arch=any&searchon=names&keywords=tinyxml2)

dirk-thomas, 2017-08-22:
> Maybe a section in the README covering this would be good?

That would be great. Can you please create a PR for this suggestion adding a `README.md` file.

dirk-thomas, 2017-09-25:
@hashworks Can you please either follow up with a PR to the README if this is still a problem or close the ticket. Thanks.

hashworks, 2017-09-26:
Sorry for the delay, kinda forgot it.

PR #82 (2017-09-26): Add README.md
Mainly to add notes about tinyxml2 versions, resolves #79.
commit 7c60b5fa (2017-09-26): Add README.md (#82)

* Add README.me

Mainly to add notes about tinyxml2 versions, resolves #79.

* Add information about patched tinyxml2 version
closed 2017-09-26 by dirk-thomas