ros/rospack #86: $rospack find       throws exception error

kamalpandey1993, 2018-01-03:
rospack find is giving runtime errors on my beaglebone board. Other rospack commands are running fine like rospack depends.
Here is the screenshot

[image]
and here is the error message
debian@beaglebone:~/catkin_ws$ rospack find
terminate called after throwing an instance of 'rospack::Exception'
  what():  error parsing manifest of package class_loader at /home/debian/ros_catkin_ws/install_isolated/share/class_loader/package.xml
Aborted

dirk-thomas, 2018-01-03:
Since the error message mentions a problem parsing the file `/home/debian/ros_catkin_ws/install_isolated/share/class_loader/package.xml` please post the content of that file.

dirk-thomas, 2018-01-03:
> i have installed ros on my board from source files as apt-get repository was unable to find it.

Without knowing what you have done we can't give you any helpful feedback. Please describe your exact steps / which tutorial you have followed and what happened when it "was unable to find it". Also this question is different from the `rospack` problem and might be better suited for answers.ros.org.

dirk-thomas, 2018-01-04:
Closing since the same question has been double posted here: https://answers.ros.org/question/278733/rospack-find-throws-exception-error/

Please provide the requested information on answers.ros.org.

closed 2018-01-04 by dirk-thomas

dirk-thomas, 2018-01-10:
See #77.