ros/rospack #92: 'boost/tr1/unordered_set.hpp' file not found

shivamkumarsingh114, 2018-08-29:
I understand that new versions of boost doesn't have tr1 file but in Kinetic-devel branch it is still included in the code which throws this error.
since lunar is still not supported for OSX I am trying to install Kinetic with No avail.
OSX: 10.13
ROS-Kinetic

[image]

dirk-thomas, 2018-08-29:
In the Boost version(s) available on the platforms targeted by Kinetic the `tr1` headers are present. `tr1` was removed on the lunar-devel branch in #80 since newer Boost version don't provide the `tr1` headers anymore.

> since lunar is still not supported for OSX I am trying to install Kinetic with No avail.

macOS is a rolling release. When installing dependencies with Homebrew you will get the latest version of Boost. As a consequence I would recommend to use the latest ROS distro.

I am not sure why you think that Lunar (or Melodic) are not supported on macOS? [This discourse post](https://discourse.ros.org/t/ros-lunar-on-os-x/2967) sounds like it should work.

dirk-thomas, 2019-01-26:
I will go ahead and close this ticket assuming my previous comment resolved the problem.

closed 2019-01-26 by dirk-thomas

GiulioPettenuzzo, 2019-03-16:
I am in the same situation, but I don't understand how to solve it!

DaHaiHuha, 2019-03-18:
I want to know the solution, what if i just download 'boost/tr1/unordered_set.hpp' file and put it the right place?

daweim0, 2019-05-08:
Did that work?

jveitchmichaelis, 2019-11-28:
There are issues with ROS Melodic on Arm64 and Kinetic is the last known good version. I've added a PR to fix this that should retain back-compatibility with previous versions of boost (the problem was only introduced with 1.65 I believe).

If this is a wontfix, then others can use my patch here: https://github.com/jveitchmichaelis/rospack/tree/kinetic-fix-boost