ros/rospack #98: Kinetic release with #97 included (ros_environment dep)

gavanderhoorn, 2019-06-04:
Giving this some more visibility. Apologies for my persistence.

From https://github.com/ros/rospack/pull/97#issuecomment-496563095:
> @dirk-thomas: would it be possible to release `rospack` for Kinetic with this patch included? I believe [2.4.4...kinetic-devel](https://github.com/ros/rospack/compare/2.4.4...kinetic-devel) shows that this is the only change missing since `2.4.4` which is the latest version of `rospack` on Kinetic.

with the release of ROS 2 Dashing out of the way, would it perhaps be possible to consider such a release?

dirk-thomas, 2019-06-08:
Closing as duplicate.

closed 2019-06-08 by dirk-thomas

gavanderhoorn, 2019-06-08:
Could you link to the issue this is a duplicate of?