ros/rospack #99: rospack depend does not resolve distribution dependencies

Vuwij, 2019-07-04:
When running
rospack depend application

Following error is retrieved
[rospack] Error: package 'application' depends on non-existent package 'opencv3' and rosdep claims that it is not a system dependency. Check the ROS_PACKAGE_PATH or try calling 'rosdep update'.

Where application depends on opencv3 as a package dependency. Shouldn't it be able to resolve this?
https://github.com/ros/rosdistro/blob/master/kinetic/distribution.yaml
It can be found here

Vuwij, 2019-07-05:
http://answers.ros.org/question/327733/installing-all-ros-dependencies-of-a-single-package-rather-than-the-entire-workspace/

dirk-thomas, 2019-07-15:
Since the original question has been marked as answered I will go ahead and close this ticket. If that is not the case please comment with more information (e.g. reproducible steps) and the ticket can be reopened.

closed 2019-07-15 by dirk-thomas