ros/rospack #100: Avoid Python linkage?

mikepurvis, 2019-07-10:
Because `rospack` links to Python, roslib, roscpp, and ultimately most of `lib/lib*.so` on a typical ROS 1 workspace links to Python. Ideally, base libraries should be python-free, with the python dependency encapsulated in an `so` in the `lib/pythonx.y/(dist|site)-packages` directory structure somewhere.

However, the actual usage here is pretty minimal— the only functions which call though to Python are:

- Rosstackage::cpp_exports
- Rosstackage::reorder_paths
- Rosstackage::isSysPackage

And none of them seem particularly dependent on doing anything interactive or "fancy" with the Python interface, that is, they could just as easily be accomplished by parsing the stdout of a Python subprocess. Would a PR which changes to this approach be considered?

dirk-thomas, 2019-08-12:
I am fine with any kind of patches as long as they pass the unit tests and are sufficiently tested manually (in case of such significant changes).

> supporting an extremely legacy rosmake/rosbuild build setup?

Another approach is to just remove these parts in the upcoming distro Noetic.

mikepurvis, 2019-08-19:
We ended up just patching out all of this functionality, and it has not negatively affected our system at all. I can see about putting up a PR for this if it would be welcome in Noetic.

dirk-thomas, 2019-11-21:
@mikepurvis Are you still interested in this or is the side-by-side installability achieved by #109 sufficient for your use case?

mikepurvis, 2019-11-22:
#109 permits Py2- and Py3-linked rospack to be installed (and fully functional) in separate Py2 and Py3 workspaces on the same machine. Further hackery was required on our side to enable roscpp-linked binaries to behave properly in a scenario where Py2- and Py3-built packages are installed into the _same_ workspace. However, I believe we're the only ones doing that, so it's probably not that big a deal at this point.

OTOH, it's definitely smelly to have basically the entirety of `lib/*.so` linking to `libpython`, particularly when the functionality in use is all vestigial and rosbuild-related at this point. We patched it out months ago and have encountered no side effects. I will put up a PR for comment if there's interest.

dirk-thomas, 2019-11-22:
> We patched it out months ago and have encountered no side effects. I will put up a PR for comment if there's interest.

Absolutely, please create a PR for it. We can always retarget the branch once we have one for Noetic.

mikepurvis, 2021-09-29:
Closing this since there's not much point now. :)

closed 2021-09-29 by mikepurvis