ros/rospack #101: Calling "mkstemp" without securely setting umask first

danielwangksu, 2019-07-10:
Using mkstemp without setting the permission mask with umask harmful.
https://github.com/ros/rospack/blob/3358d21fbeb5355a459e94354c2aa95e0a376e57/src/rospack.cpp#L2064

dirk-thomas, 2019-07-15:
Please consider to provide a pull request for this.

dirk-thomas, 2019-07-18:
Seeing the proposed patch can you clarify why you consider the `mkstemp` call without a `umask` to be *harmful*?

danielwangksu, 2019-07-18:
The POSIX specification of mkstemp() does not say anything about file modes, although newer version of glibc make file modes changes specifically. It is not necessary, but I think it is probably better for other potential platform to make sure its file mode creation mask (umask(2)) is set appropriately before calling mkstemp() (and mkostemp())

dirk-thomas, 2019-07-18:
I am trying to understand where do you see writing the file with even 0666 being *harmful*?

danielwangksu, 2019-07-18:
With 666 any process can read and write to the file. Could be a potential way to inject harmful data

closed 2019-08-12 by dirk-thomas