ros/rospack #117: rospack cache files created with 000 permissions

ph1l, 2020-05-07:
The permissions on these files make no sense. Why write a file that no one can read?

```
phil@marble-XPS-15-9560:~$ cp -a .ros/ /tmp/ros_backup
cp: cannot open '.ros/rospack_cache_02946445166995464026' for reading: Permission denied
phil@marble-XPS-15-9560:~$ ls -lah .ros/rospack_cache_*
---------- 1 phil phil 6.0K Mar  3 12:07 .ros/rospack_cache_02946445166995464026
```

https://answers.ros.org/question/351675/rospack-cache-files-in-ros-directory-with-no-access-rights/

ph1l, 2020-05-07:
It looks like the package build for melodic succeeded, but there are no artifacts in jenkins. Anywhere I can d/l then to test?

dirk-thomas, 2020-05-12:
There are no automatic binary artifacts. The Jenkins build only compiled the package and ran its tests.

dirk-thomas, 2020-05-12:
Please build the package with your changes from source locally and test that the patch actually adresses the problem (which the first version of the PR doesn't seem to do).

ph1l, 2020-05-12:
looks like the call to `mkstemp` (https://github.com/ros/rospack/blob/noetic-devel/src/rospack.cpp#L2080) should be creating the cache file, right? Am I misreading the man pages for `mkstemp` and `fdopen`?

dirk-thomas, 2020-05-13:
This is actually a regression since #106. Without the `umask` calls the file is created with `600` for me.

I am looking into a proper fix...

dirk-thomas, 2020-05-13:
Please try #119 which should fix the problem.

PR #119 (2020-05-13): fix umask for rospack cache files
Fixes #117. The regression was introduced in #106.

This needs to be backported to `melodic-devel`.
closed 2020-05-13 by dirk-thomas