ros/rospack #120: Depends incorrectly fails for 'non-existant' package libuvc

StefanFabian, 2020-06-25:
For the package `libuvc_camera`, rospack depends fails with the message:
```
[rospack] Error: package 'libuvc_camera' depends on non-existent package 'libuvc' and rosdep claims that it is not a system dependency. Check the ROS_PACKAGE_PATH or try calling 'rosdep update'
```

But `rosdep resolve libuvc` works as expected:
```
#apt
ros-melodic-libuvc
```

Any idea what could cause this?

dirk-thomas, 2020-06-30:
I am not sure if you have seen the [ROS support guidelines](http://wiki.ros.org/Support) but we kindly ask to raise questions like this on [answers.ros.org](http://answers.ros.org/) instead.

The rational for asking all kind of questions in a single place is:
* There are many more people reading questions there so your chances on getting an answer and in a timely manner are much higher.
* In the future other users will search there for similar problems and can find your question and the potential answers.
* The issue tracker in this repo is used to track bugs, feature requests, etc. But your question is not such an action item.

Therefore I will close this ticket here for now. If you have asked your question on answers.ros.org please feel free to add a link to the question to this ticket so that future readers can find the related question.

closed 2020-06-30 by dirk-thomas

StefanFabian, 2020-07-01:
I respectfully disagree with this assessment.
This is not a question on how to use this but whether this is intended behavior or a bug.
I would be leaning towards the latter because the statement that rosdep claims it is not a system dependency is clearly false unless system dependency means something other than an apt package is available for that key.
`libuvc_camera` is just an example of a commonly used package for which this happens.

dirk-thomas, 2020-07-06:
> This is not a question on how to use this but whether this is intended behavior or a bug.

The behavior looks correct to me. Both statements `rospack` makes are correct:

1. package 'libuvc_camera' depends on non-existent package 'libuvc'

   `libuvc` is not a "correct" ROS package. Checking the content of the Debian package `ros-melodic-libuvc` shows that it does *not* contain a `package.xml` file.

   So the problem needs to be addressed in the `libuvc` package itself. Please consider to reporting the problem in the upstream repo: https://github.com/ktossell/libuvc.git

2. rosdep claims that it is not a system dependency

   Which is also true since there is no `rosdep` key `libuvc` which maps to a Debian package provided by Ubuntu.

> system dependency means something other than an apt package is available for that key.

A system dependency for `rosdep` is a mapping of a `rosdep` key to a package provided by the Ubuntu / Debian / whatever distribution. While ROS packages are built into a Debian package and afterwards available via apt with the prefix `ros-<distro>-` they are not "system dependencies".

StefanFabian, 2020-07-07:
Ah okay, thank you for the clarification!